sparkfun/SparkFun_LSM9DS1_Arduino_Library #22: Initializing IMU breaks PWM on raspberry pi

saspelund, 2019-04-04:
I am working with a Raspberry Pi 3B. I am trying to use this IMU and provided libraries/code in a robot that also utilizes PWM for motor control. I can show (LED+oscilloscope) that the PWM (using wiringPi) works immediately before but not after the following code:

```
LSM9DS1 imu(IMU_MODE_I2C, 0x6b, 0x1e);
imu.begin();
```

If the PWM is connected to PWM0 (either pin), then the measured PWM signals jump all over the place rather than smoothly increasing the duty ratio and then decreasing it multiple times. If the PWM is connected to PWM1, then the PWM signal remains flatlined.

The IMU works fine which is very nice. Thank you!

Similar issue: https://raspberrypi.stackexchange.com/questions/53455/pwm-conflicts-with-i2c

santaimpersonator, 2019-04-08:
I am a little confused to your issue/setup as this is repository is for an **Arduino**> library.

saspelund, 2019-04-11:
It turns out that I was actually using a c++ wrapper from https://github.com/akimach/LSM9DS1_RaspberryPi_Library and it was using a different pin mapping than my code.

Thanks for the response!

closed 2019-04-11 by saspelund