sparkfun/SparkFun_LSM9DS1_Arduino_Library #24: power consumption

savan8192, 2019-10-23:
Hi everyone,

I have been playing with LSM9DS1 for quite some time. Now I want to reduce its power consumption.

Currently ardunioClone + LSM9DS1 draws about 60mA. If I disabled gyro, mag and two of the axis of Accelerometer and the current draw has gone down to 35-40mA. On disconnecting the LSM9DS1 from arduino entirely, the same sketch consumes 2-10 mA.

LSM9DS1 should draw typically 4.5mA. Any idea where I am going wrong?

I have connected VDD, GND, SDA, and SCL directly to the same Arduino pins.

Thanks in advance

savan8192, 2019-10-23:
Sketch here - https://github.com/sparkfun/SparkFun_LSM9DS1_Arduino_Library/blob/master/examples/LSM9DS1_Basic_I2C/LSM9DS1_Basic_I2C.ino - consumes about 49mA.

santaimpersonator, 2019-12-27:
Hi there, it sounds like you are looking for [technical assistance](https://www.sparkfun.com/technical_assistance). Unfortunately, this is channel for issues specifically with the hardware. Please feel free to post a topic in our [forums](https://forum.sparkfun.com/). Our technical support team will do their best to assist you.

closed 2019-12-27 by santaimpersonator