sparkfun/SparkFun_LSM9DS1_Arduino_Library #34: Interrupt configuration pushPull logic broken

choller, 2020-12-15:
Currently, the interrupt configuration function looks like this:

```
void LSM9DS1::configInt(interrupt_select interrupt, uint8_t generator,
	                     h_lactive activeLow, pp_od pushPull)
{
	[...]

	if (pushPull) temp &= ~(1<<4);
	else temp |= (1<<4);
	
	[...]
}
```

with `pp_od` being declared as

```
enum pp_od
{
	INT_PUSH_PULL,
	INT_OPEN_DRAIN
};
```

The LSM9DS1 spec says:

```
PP_OD   Push-pull/open-drain selection on the INT1_A/G pin and INT2_A/G pin.
        Default value: 0
        (0: push-pull mode; 1: open-drain mode)
```

So with the current code, if you specify INT_PUSH_PULL (which is 0), it will write 1 into the register and switch to open-drain mode instead.

The parameter should be renamed and the logic swapped to give the correct result.

Fwiw, this means that all uses of interrupts with this library are broken.

CAP1Sup, 2022-04-30:
@choller could you write a PR [on my fork](https://github.com/CAP1Sup/Arduino_LSM9DS1)? I'd be happy to merge it

saeidb71, 2022-09-30:
Hello. It seems that interrupts are not working correctly. I wonder if it is because of the issue mentioned here. Are you going to update the code?

CAP1Sup, 2022-10-03:
Unfortunately... no. I really don't have the time to put into it. If you want to make a PR, I would be happy to merge it